DCC-Lab/RayTracing #12: Overlapping labels

dccote, 2019-02-15:
It is not easy to put several labels on a graph without any overlap.  A solution should be found.

dccote, 2019-04-08:
I think the best option would be to make a small class that handles all labels in a given graph.

dccote, 2020-05-14:
@FrancoisCoteIng : Ludovick and I are discussing a new design for managing graphs.  Did you get anything done on overlapping labels?  We could take care of it in the process.

FrancoisCoteIng, 2020-05-14:
j’ai travaillé un peu dessus, la seule facon que j’avais trouvé, cétait de bouger up and down le label si 2 points focaux se chevauchaient à l’intérieur d’un certain pourcentage de la taille du plot. ca avait dlair de bien marché et c’était 2 ou 3 lignes à ajouté. mais cest seulement pour les points focaux, pas pour tous les labels.

J’avais besoin de travailler un peu dessus avant de faire un pull request, la je travaillais sur le VMS du cervo. mais en ce moment je suis bloqué alors j’allais retravailler dessus.

C’était quoi votre idée? Une classe à part? Mon idée fonctionne, mais a ses limites.

closed 2020-07-07 by JLBegin