DCC-Lab/RayTracing #45: When displaying a graph, ctrl-c does not quit the python script

dccote, 2020-05-05:
On macOS and probably Windows, if a python script displays() a trace, then Ctrl-C will not quit the program.  One must close the window.  This is very annoying but it is not a problem specific to RayTracing, it is a problem related to matplotlib.  Many hacks exist on the web to solve that problem, but I have not found one that is robust and working well (you can google `matplotlib ctrl-c`).

I want Ctrl-C to work, but I want I solution that works on all platforms.

antho214, 2020-05-06:
I found this [issue](https://github.com/matplotlib/matplotlib/issues/6897) on the matplotlib library. The `ctrl-c` issue seems to be related to the backend used by matplotlib.

On macOS, the default backend used is MacOSX, while on Linux the backend uses Qt5Agg
```python 
# From macOS
from matplotlib import pyplot 
pyplot.get_backend()
>>> MacOSX

# From Linux/Ubuntu
from matplotlib import pyplot 
pyplot.get_backend()
>>> QtAgg5
```
Forcing the backend on macOS to `Qt4Agg` or `Qt5Agg` solves the `ctrl-c` issue. 
```
import matplotlib 
matplotlib.use("Qt5Agg")
```
Unfortunately, I cannot test the solution on Windows. 

__Notes__
- This solution requires Qt4/5 to be installed on the computer.
- Slightly changes the `matplotlib.figure` interface on macOSX.

GabGabG, 2020-05-06:
> I found this [issue](https://github.com/matplotlib/matplotlib/issues/6897) on the matplotlib library. The `ctrl-c` issue seems to be related to the backend used by matplotlib.
> 
> On macOS, the default backend used is MacOSX, while on Linux the backend uses Qt5Agg
> 
> ```python
> # From macOS
> from matplotlib import pyplot 
> pyplot.get_backend()
> >>> MacOSX
> 
> # From Linux/Ubuntu
> from matplotlib import pyplot 
> pyplot.get_backend()
> >>> QtAgg5
> ```
> 
> Forcing the backend on macOS to `Qt4Agg` or `Qt5Agg` solves the `ctrl-c` issue.
> 
> ```python
> import matplotlib 
> matplotlib.use("Qt5Agg")
> ```
> 
> Unfortunately, I cannot test the solution on Windows.
> 
> **Notes**
> 
> * This solution requires Qt4/5 to be installed on the computer.
> * Slightly changes the `matplotlib.figure` interface on macOSX.

I tried it on Windows and it doesn't work. There is a way to close the window on CTRL+C event that doesn't require Qt4/5. We can bind an event to the figure. For example:
```python
import matplotlib.pyplot as plt

def handle_close(evt):
    if evt == "ctrl+c":
        plt.close() # close the figure
        exit() # exit the program, or raise KeyboardInterrupt for interrupt stack

fig = plt.figure()
fig.canvas.mpl_connect('key_press_event', handle_close)
plt.show()
```

dccote, 2020-05-06:
Two things: 
@antho214 I really don't want to be dependent on such a large library just for ctrl-c.  
@GabGabG The code does not close the window when ctrl-c is typed from the script window.  Ctrl-C is a signal in the Unix/Windows terminology.  It calls an interrupt function that will execute regardless of the code currently running.  A good place to read about it is  https://docs.python.org/3/library/signal.html

*However*, matplotlib seems to intercept these signals and somehow prevents our routine from being called. It really is a matplotlib issue and it appears to be fairly common on the Web.
I still don't have a solution.

closed 2020-05-07 by dccote