DCC-Lab/RayTracing #49: Evaluate various documentation formats, make a choice

dccote, 2020-05-06:
The documentation is currently based on simple docstrings.  We need better by looking at Sphynx and other options. Ideally, the format will automatically produce a nice looking HTML or Markdown output.

JLBegin, 2020-05-13:
Similar to scikit-learn, I think we should have a typical API reference documentation (auto-generated from doctrings) and a user guide (including various demos). 

Regarding the API reference, the doctrings should follow a standard format like [Numpy](https://sphinxcontrib-napoleon.readthedocs.io/en/latest/example_numpy.html#example-numpy) and include a description, the inputs, the outputs and some examples. 

The HTML is easily generated by [Sphinx](https://www.sphinx-doc.org/en/master/) using reStructuredText code.

JLBegin, 2020-05-13:
I think this issue should be merged with #30

PyMarc2, 2020-05-13:
+1 for the use of Sphinx from reStrcutu. I think it is the conventionnal and most robust way to produce documentation in python since the guidelines are clear in the doc. I don't think it should be a difficult choice.

closed 2020-05-19 by dccote