DCC-Lab/RayTracing #52: Unit testing for near 100% coverage needed

dccote, 2020-05-06:
Best places to start: `Matrix`, then `MatrixGroup`, `OpticalPath`, then `ImagingPath`

Unit tests are very limited at this point.  To create more tests, do to the tests directory and create a file with a name starting with `testsWhatever.py`. Look at other files for examples.
To run, simply use: `python -m unittest testsWhatever.py`.
To run with coverage, use: `coverage run -m unittest discover`, then `coverage html` and look at `coverage_html_report/index.html`.

GabGabG, 2020-05-08:
@dccote Is there a reason why there are 3 different files to test matrix? matrixTests.py, testMatrix.py and testsMatrix.py

JLBegin, 2020-05-12:
@GabGabG just to let you know that I'm currently debugging and testing `MatrixGroup.largestDiameter()`, `ImagingPath.displayRange` (new) and `Matrix.AxesToDataScaling()` in another branch called `display`.

dccote, 2020-05-19:
Replaced by smaller issues specific to files.

closed 2020-05-19 by dccote