DCC-Lab/RayTracing #53: Figure out what the accepted nomenclature is for chief ray, axial ray, marginal rays

dccote, 2020-05-06:
There is a chief ray for every point on the object.  However, *the* chief ray from the point on the object at the edge of the field of view is often referred to as "the chief ray". We may want to call it the System chief ray but I am not sure if this is correct
Same with marginal rays: there are marginal rays from all points, but the useful one is the one on axis at the object.  We need to determine what the correct nomenclature is by checking in books:
Kloos, Saleh Teich, Smith, etc... and people in the field.

dccote, 2020-05-12:
1. Gerhard Kloos calls the axial and principal rays the two rays for an optical system.  The principal ray is the chief ray from the margin of the object. The axial ray is the ray from the axis that touches the margin of the aperture stop.
2. Many people on the web define chief ray and principal ray to be synonymous.
3. Some call the axial ray the paraxial marginal ray.http://ecee.colorado.edu/~ecen5616/WebMaterial/09%20apertures%20and%20stops.pdf
4. My discussions with Simon Thibault says "marginal ray of the system" and "principal ray of the system"
For now, I think the best option is 1.  Marginal rays are not necessarily axial and for sure there is always a chief ray for any point (the chief ray definition does not appear to be ambiguous).

dccote, 2020-05-27:
@Valeriouus Please take note of nomenclature for article.

Valeriouus, 2020-05-27:
We decided for the article that : 
- principal ray starts at the top edge of the object and passes through the center of the optical axis at the AS plane.
- chief ray defines any ray still passing though the center of the AS, but starting at any point on the object
- axial ray starts on the axis at the object plane and propagates through the AS at the maximal possible height
- marginal ray defines any ray passing at the maximal possible height of the AS, regardless of the starting point on the object

closed 2020-05-27 by Valeriouus