DCC-Lab/RayTracing #55: Article has no clear outline and sections

dccote, 2020-05-06:
Those interested in writing need to meet with me to make a quick outline and split the work of writing some of those sections.

dccote, 2020-05-06:
I am suggesting @shadimasoumi @ElaheParham and @Valeriouus to work on this.

shadimasoumi, 2020-05-06:
I confirm, and have you chosen a Journal for publishing the article?
Knowing this will help in writing! (I mean, does it need a short or a long introduction, discussion and etc…)

shadimasoumi, 2020-05-06:
> I am suggesting @shadimasoumi @ElaheParham and @Valeriouus to work on this.
> Please confirm.

I confirm, and have you chosen a Journal for publishing the article?
Knowing this will help in writing! (I mean, does it need a short or a long introduction, discussion and etc…)

dccote, 2020-05-06:
I am thinking of sending it to Journal of Neurophotonics for several reasons (they like tutorials and I am a topical editor therefore I like the journal).  Also, I think this is a good target audience but we can still publicize it outside.

closed 2020-05-26 by dccote