DCC-Lab/RayTracing #62: Python requirements and version compatibility is not clear

JLBegin, 2020-05-08:
## RayTracing requires tkinter
If you are not installing Python with Anaconda, it is important to keep tcl/tk support during python install since it is required by matplotlib's backend (tkinter). This is no problem if you follow default installation options. 
[image]

## Version compatibility
I tested RayTracing and it **currently works on Python 3.6 and newer versions (3.7 & 3.8)**. 

Since Python 3.5 is still an [active Python 3 release](https://www.python.org/downloads/), I tried to install RayTracing, but it fails to run because of a new type hint syntax for variable annotations ([PEP 526](https://www.python.org/dev/peps/pep-0526/)) used in imagingpath.py on line 87 & 88:
```python
thetaUp: float = (stopDiameter / 2.0 - A * y) / B
thetaDown: float = (-stopDiameter / 2.0 - A * y) / B
```
I tested with the following edit on 3.5 which fixes the issue and works for 3.5+.
```python
thetaUp = (stopDiameter / 2.0 - A * y) / B  # type: float
thetaDown = (-stopDiameter / 2.0 - A * y) / B  # type: float
```
## Conclusion
- We need to choose if we want to **discard [PEP 526](https://www.python.org/dev/peps/pep-0526/) to support Python 3.5+ or keep it and 'only' support 3.6+**. 
- `README.md` needs to be updated to tell the required python version.

JLBegin, 2020-05-08:
I guess the tcl/tk requirement is a no-brainer, but if someone gets an `ImportError: No module named tkinter` then this is how you fix it.

dccote, 2020-05-10:
I really want type hints.  I suggest we recommend 3.6 and above, unless there are very good reasons to limit to 3.5.
Type hints are very important for educational purposes and clarity.  It is one of the design goal, so we should not compromise on this considering it is only a minor version 3.5->3.6 requirement. I think we need to modify setup.py to reflect that.

closed 2020-05-11 by dccote