DCC-Lab/RayTracing #70: The README that gets published on PyPi does not have images

dccote, 2020-05-12:
If you look up the PyPi raytracing page, the figures don't show up.
I think the links to the images in the README.md should be made absolute, and directly to the master branch on GitHub.

dccote, 2020-05-13:
I used absolute URLs, but also when linking to GitHub files, we must use `raw` not `blob`. I had to use "master" in the link, so we will only be able to confirm this is working fine once it is merged in master.

PyMarc2, 2020-05-15:
>This is the only thing that worked for me. In markdown, a link would look like this: 
>[image]

This response is from : https://stackoverflow.com/questions/41983209/how-do-i-add-images-to-a-pypi-readme-that-works-on-github

Someone has to change this manually I suppose. I could do it.

dccote, 2020-05-22:
Sovled with separate pull request #143

closed 2020-05-22 by dccote