DCC-Lab/RayTracing #79: The __main__ file is not particularly easy to read and is hard to maintain

dccote, 2020-05-13:
It would be nice and useful to have a function, in some sort of `PlotManager`, that would take code as an argument, run it, and print the code on screen with syntax highlighting. We could even have have the option of providing the code as a string or as a file on disk. This way, for instance, we could store all examples in the examples directory and access them directly.

PyMarc2, 2020-05-13:
Could we just add arguments to the argparse for the file and code management?
`python -m raytracing -f testRaytracing.py`
`python -m raytracing -s "ALL THE CODE AS A STRING with \n and stuff."`

and for the `python -m raytracing -e`, I agree that every example should be in the examples folder and a sort of `ExampleManager` would select the example to display from the file. Maybe even this object could manage the input from the argparse when we use -f or -s for example @dccote

PyMarc2, 2020-05-13:
However, at this point, if they have a code `test.py` that look like this:
`
path = ImagingPath()
    path.label = "Demo #1: lens f = 5cm, infinite diameter"
    path.append(Space(d=10))
    path.append(Lens(f=5))
    path.append(Space(d=10))
    path.display()
`
At this point, isn't easier AND clearer to just IMPORT the raytracing library components into their code `
test.py` ?

PyMarc2, 2020-05-13:
I think that segmentation of the example codes is useful: less messy and more easily understandable if you want to make a new example, you just add it in the example folder

PyMarc2, 2020-05-13:
ok. Here's the plan
Each example.py file is separated. The example comment is in docstrign instead.
There will be a ExampleManager which will generate the figure and save it in pdf.

Then, the code of that file will the converted to coloured HTML using the pygment module.
From the HTML output we will generate a PDF. Then we will juxtapose the 2 PDFs one over the other in a new PDF which will be displayed. Thus, the code will be highlighted and selectable.

PyMarc2, 2020-05-13:
See branch #issue79-mainEnhancement

PyMarc2, 2020-05-14:
[image]
This is the HTML file output for the code highlighting and selectability

PyMarc2, 2020-05-15:
The output seems leggit. Here's a quick view. Does it asthetically please you @dccote ?
[image]

dccote, 2021-05-05:
I started a new branch to work on this.  Using `pygments`, we can create a bitmap with the code and syntax-coloring... so we should create a bitmap, and show it in a matplotlib window.

closed 2021-05-06 by dccote