DCC-Lab/RayTracing #80: Tests stop when warning is issued

dccote, 2020-05-13:
The warnings package now makes the tests stop if a warning was issued.
There is a file called `env.py` that is imported in all tests.  There is a setting for the behaviour of warnings that can be set, and it should be added to `env.py`.

GabGabG, 2020-05-13:
@dccote when I run ``python -m unittest`` in my terminal (with all test files), it also stops but not because of warnings. It stops because of tests in testsAsynchrnousTesting and testsMatplotlib.
[image]

But when I remove said files, it works normally, even with warnings:
[image]

Can you run ``python -m unittest`` on my new branch fixWarnings? I only removed testsAsynchrnousTesting and testsMatplotlib and I would like to know if it works normally on your computer, because it does on mine.

dccote, 2020-05-18:
> @dccote when I run `python -m unittest` in my terminal (with all test files), it also stops but not because of warnings. It stops because of tests in testsAsynchrnousTesting and testsMatplotlib.

You are 100% right. I merged master into your branch, fixed the conflates, and will merge back into master.

dccote, 2020-05-20:
Was solved.

closed 2020-05-20 by dccote