DCC-Lab/RayTracing #91: Empty MatrixGroup raises IndexError in largestDiameter()

GabGabG, 2020-05-14:
When we create a `MatrixGroup` with no element, `largestDiameter() ` raises an IndexError exception because we try to access the first element of an empty list. I think this should be handled in a better way.

Question:
Should we keep it like that?
 - Yes: Ok.
 - No: Should we throw another type of exception, more specific with a proper message? Or should we just return None (or another thing)?

dccote, 2020-05-18:
I think the answer is that no elements in a group means the largestDiameter is float("+inf"). We should return that.

GabGabG, 2020-05-19:
`largestDiameter()` now returns 0 when matrix group is empty. @JLBegin is there a reason why it is 0 or is it arbitrary? If we change it to infinity, will it mess with what you have done?

JLBegin, 2020-05-19:
> `largestDiameter()` now returns 0 when matrix group is empty. @JLBegin is there a reason why it is 0 or is it arbitrary? If we change it to infinity, will it mess with what you have done?

arbitrary. infinity also works on my end

PR #118 (2020-05-19): Empty matrixGroup now returns 'infinity' as largest diameter
When a matrix group is empty (no matrix/component), its largest diameter is infinity (no aperture). Fixes #91 .
closed 2020-05-19 by dccote