DCC-Lab/RayTracing #103: Automatic class hierarchy would be useful

dccote, 2020-05-18:
I like using webgraphviz to visualize hierarchies.  I made a graph in __init__ but it must be updated manually.  It would be nice to have the class hierarchy automatically.  We could put the code in docs/ or examples/ not sure where. This would be for developers but maybe also users.

dccote, 2020-05-19:
This is a fairly good solution that needs a tiny bit of work:

```python
def allChilds(aClass):
    for child in aClass.__subclasses__():
        print("\"{0}\" -> \"{1}\"".format(aClass.__name__, child.__name__))
        allChilds(child)

if __name__ == "__main__":
    print("digraph G {")
    print("  rankdir=\"LR\";")
    allChilds(Matrix)
    print("}")
```

dccote, 2020-05-22:
Make sure it gets updated whenever we create new documentation, use other formats if needed, so we can use only Python.  graphviz requires separate, non-python package.

PyMarc2, 2020-05-22:
Yes. Do you want it to change the .png in the README.md also, or only in the read-the-docs? There's a sphinx extension I found to build the graph, I'll verify if I can output the .png from sphinx also

dccote, 2020-07-17:
A simple function was added to utils.py for now.

closed 2020-07-17 by dccote