DCC-Lab/RayTracing #119: Unused imports

bopaquin, 2020-05-19:
While looking at my IDE and I've got hundreds of unused import reports.

dccote, 2020-05-19:
Be specific so we can identify exactly what is meant here and reproduce/validate. Maybe paste the list of identified modules?

dccote, 2020-05-19:
> [Linting - pylint.txt](https://github.com/DCC-Lab/RayTracing/files/4653336/Linting.-.pylint.txt)

A better description would describe how the pylon.txt was generated so we can repeat it on our computers.

`pylint ... blablalba`

dccote, 2020-05-19:
Ok, this is definitely a low priority issue because we still have some bugs and some documentation issues weblike to work on before the publication. we can keep it, but it is a fairly tedious process.  Maybe we could do it on a few key files (such as `matrix.py` and `ray.py` for instance), but still, at a later time.

dccote, 2020-05-20:
There is an Action (automatic script) that could run pylint every time someone pushes code.  **Once we have the code cleaned up**, we could install that.
https://github.com/marketplace/actions/github-action-for-pylint

dccote, 2021-05-02:
This and #388 should be worked on together.

closed 2021-09-16 by github-actions[bot]