DCC-Lab/RayTracing #132: rayCountHistogram() of Rays raises exception when already computed

GabGabG, 2020-05-20:
When we compute `rayCountHistogram()` for a first time, it works perfectly. But when we run it a second time, it throws an exception because a variable is referenced before asignment.

GabGabG, 2020-05-20:
From what I understand: we cache _yHistogram, but we don't cache the bins, so this is the main problem. But it's not all: if we compute the histogram with, let's say 10 bins, we chache histogram values, then if we want to compute histogram with another number of bins, it won't work because we have cached values. We could keep a parameter about the histogram properties (nb bins, min and max). We should also chache the bin edges, or the xvalues, which are returned at the end and cause the initial issue of the exception.

dccote, 2020-05-20:
I am not sure anymore that caching the values is that critical, and I hate that it adds a bit of complexity to the process (like the one here!).  That being said, super happy it is 1) noticed 2) solved.

PR #133 (2020-05-20): Fixed problems with Rays histogram methods
When we computed `rayCountHistogram()` and `rayAnglesHistogram()` for the first time, it was working perfectly, but when we tried it a second time, it was not working. See issue #132 for more info on why it was not working.

I added new intern variables related to histogram computation, nothing major.

Fixes #132
closed 2020-05-20 by dccote