DCC-Lab/RayTracing #144: Matrix.display() does not work for default value

ElaheParham, 2020-05-22:
I wrote the following code to run an example for Matrix.display()

```
from raytracing import *
M= Matrix(A=1,B=3,C=1,D=4)
M.display()
```

It does not display anything and it does not rais any error. 
when I add other inputs of the matrix It works properly.

```
from raytracing import *

M1= Matrix(A=1,B=3,C=-1/3,D=1,physicalLength=3,frontVertex=4,
           backVertex=2,frontIndex=1,backIndex=1,apertureDiameter=6,
           label='Lens')
M1.display()
```

I suggest to add some errors or change the default value so it works!

ElaheParham, 2020-05-22:
The main problem is with the frontVertex and backVertex values that are defined None

dccote, 2020-05-22:
Good point: front and back vertices really represent actual physical interfaces, so it is possible to not have any (Space(d=10) is an example of such matrix).  Constructing a matrix manually is possible, but it needs to fill certain requirements: the determinant must be at least n1/n2 (or the inverse, I don't remember), and if C != 0, then vertices must be defined for sure  because C is the power (-1/f) and therefore there must be vertices. If these conditions are not filled, we should raise an ValueError()

ElaheParham, 2020-05-22:
> Good point: front and back vertices really represent actual physical interfaces, so it is possible to not have any (Space(d=10) is an example of such matrix). Constructing a matrix manually is [...]

I will add that to the details of the function display()

dccote, 2020-05-23:
One thing we should definitely do is check the determinant in Matrix.__init__() to confirm that it makes sense, and raise an exception if it does not: we have everything we need because `A*D-B*C = n1/n2`.

In display() we should find an intelligent way to `display` if there is not much to display. Maybe going through points of interest?  Maybe only do something if C != 0? There is not much to do if there is no power, but some systems have C == 0 total, but C != 0 in between.

ElaheParham, 2020-05-25:
So, an exception needs to be added to the Matrix.init to check the validity of the defined matrix.
Also, instead of "None" as the default value for the frontVertex and backVertex i think the position of the surfaces of the object is a good choice.

I wanted to comment about the frontVertex and backVertex value which are corrently the position of the surfaces to the principal planes. I think it would be better if instead of position, they are the distance between front surface to Pf and back surface to Pb. In this case the initial value for them can be defined zero.

ElaheParham, 2020-05-25:
> 
> In display() we should find an intelligent way to `display` if there is not much to display. Maybe going through points of interest? Maybe only do something if C != 0? There is not much to do if there is no power, but some systems have C == 0 total, but C != 0 in between.

I think It would be a good idea to divide the display in two cases. First, when there is no power and the second is when there is power.
In the first case the plot can be a box with a thickness and no more info!

JLBegin, 2020-06-16:
N.B.: The code for Matrix.display() is now under raytracing.figure.MatrixGraphic.display()

closed 2020-08-25 by dccote