DCC-Lab/RayTracing #147: Matrix.lagrangeInvariant does not work with ABCD matrixes which have physicalLength>0

ElaheParham, 2020-05-22:
I tried the following example for the lagrangeInvariant function in the Matrix class.

```
from raytracing import *
# M is an ABCD matrix of a lens (f=10)
M= Matrix(A=1,B=0,C=-1/10,D=1,physicalLength=5,label='Lens')
# R1 and R2 are two rays
R1=Ray(y=5,theta=20)
R2=Ray(y=2,theta=10)
LagrangeInv=M.lagrangeInvariant(R1,R2)
print('The lagrange invariant value for R1 and R2 is:', LagrangeInv)
```

When I run the code, I get the following error:
> TypeError: Subclass of non-null physical length must override transferMatrix()

in the example the physical length is 5, if I remove the physicalLength (or set it to zero) there will be no error and it works.

```
from raytracing import *
# M is an ABCD matrix of a lens (f=10)
M= Matrix(A=1,B=0,C=-1/10,D=1,physicalLength=0,label='Lens')
# R1 and R2 are two rays
R1=Ray(y=5,theta=20)
R2=Ray(y=2,theta=10)
LagrangeInv=M.lagrangeInvariant(R1,R2)
print('The lagrange invariant value for R1 and R2 is:', LagrangeInv)
```

and the output will be:
>>> The lagrange invariant value for R1 and R2 is: -10.0

dccote, 2020-05-23:
I need to double check the code, a find my LagrangeInvariant function pretty clumsy.

ElaheParham, 2020-05-25:
Since this value should be constant for any 2 rays, I don't know why the physical length of the element is making such error.

closed 2020-07-06 by dccote