DCC-Lab/RayTracing #149: Matrix.pointsOfInterest Returns Null

ElaheParham, 2020-05-23:
The following is the definition of this function and it returns nothing!

```
    def pointsOfInterest(self, z):
        """ Any points of interest for this matrix (focal points,
        principal planes etc...)
        """

        # ptsOfInterest = []
        # if self.frontVertex is not None:
        #     ptsOfInterest.append({'z': self.frontVertex, 'label': '$V_f$'})
        # if self.backVertex is not None:
        #     ptsOfInterest.append({'z': self.backVertex, 'label': '$V_b$'})

        return []
```

dccote, 2020-05-23:
We have to go through all elements, but that is because pointsOfInterest may not always exist: they are suppose to be "highlightable" on a diagram, so for instance, free space does not have anything of interest, and it will return an empty list. `Matrix` being the base class, by default it returns nothing.  Subclasses (such as `Lens`) will override and provide, for instance, the focal points.  I am not sure if we should provide vertices (like I obviously did originally, not sure why I removed it).

We should have a `pointsOfInterest` and `planesOfInterest` to separate both: this is really for diagram purposes. When drawing, we put points on these points, and we draw planes on the planes (principal planes for instance).

dccote, 2020-05-23:
We should have a @property pointsOfInterest and @property planesOfInterest to separate both: this is really for diagram purposes. When drawing with `display()`, we draw circles on these points, and we draw lines on the planes (principal planes for instance).  Each element could fill these dictionaries in the __init__ ?

* Space: no points of interest, no planes of interest
* Lens: 2 focal points, 2 vertices (may overlap), two focal planes and two principal planes
* Thicklens: same
* Dielectric Interface: 2 focal points (different distances), 1 vertex, 2 focal planes  and 2 principal plane
* Dielectric slab: no focal points, 2 vertices,
* Aperture: no points of interest, no planes of interest

dccote, 2020-05-24:
I just thought of this: we could probably generate those automatically in the base class by looking at self.C and backVertex, frontVertex. I will take a look.

ElaheParham, 2020-06-11:
I expected the function to print some info for me!
The function can be written as a switch/case, and when an optical path is an input, it can go through all elements (except free spaces) and print the info you mentioned.
Otherwise, if it is just a Freespace something like "Freespace has no points/plains of interest can be printed"

dccote, 2020-07-17:
I would prefer to leave any "user-facing" output outside of these functions, unless we create a new function such as "report()".

ElaheParham, 2020-07-17:
Ok! If I understood correctly this report function can work for the free space. what about the pointsOfInterest function for the elements that have points/planes of interest?

closed 2020-12-14 by dccote