DCC-Lab/RayTracing #186: Generating unnecessary images

Valeriouus, 2020-05-27:
illumination = ImagingPath()

illumination.append(Space(d=10))
illumination.append(Lens(f=10, diameter=100, label="Collector"))
illumination.append(Space(d=10+30))
illumination.append(Lens(f=30, diameter=100, label="Condenser"))
illumination.append(Space(d=30+7))
illumination.append(Lens(f=7, diameter=100, label="Objective"))
illumination.append(Space(d=7+30))
illumination.append(Lens(f=30, diameter=100, label="Tube"))
illumination.append(Space(d=30+10))
illumination.append(Lens(f=10, diameter=100, label="Eyepiece"))
illumination.append(Space(d=10))
illumination.append(Aperture(diameter=20, label="Eye entrance"))
illumination.append(Space(d=10))

illumination.display(limitObjectToFieldOfView=True, onlyChiefAndMarginalRays=True)

Gives this : 

[image]

Because of this : 

[image]

dccote, 2020-06-11:
This has been fixed. @JLBegin, please indicate which PR # fixed this and close this issue.

JLBegin, 2020-06-11:
This has not been fix yet. I will push a fix where displayRange ignores height of images if out of display.

PR #259 (2020-06-11): Fix issue #186: ignore images out of the optical path in displayRange
Fixes #186
closed 2020-06-11 by dccote