DCC-Lab/RayTracing #193: ImagingPath: chiefRay returns y=inf and theta=-inf with really big (but not infinite) aperture and no specific y given

GabGabG, 2020-05-28:
When we create an `ImagingPath` with a really big aperture stop (but not infinite), the chief ray obtained has y = -theta = inf. This doesn't really make sense.

This happens because the aperture is bigger than the max height and at one point (in `fieldOfView`), if `abs(y) < self.maxHeight`, we return `float("+inf")`. If the aperture is bigger than the max height accepted, then we should consider it like an infite aperture?

GabGabG, 2020-05-28:
If we consider it infinite, then we should be coherent and also consider it infinite everywhere, not only in specific methods, like `fieldOfView` and `fieldStop`. For example, `entrancePupil` and `apertureStop` don't consider the maximum height. Maybe we could override `hasFiniteApertureDiameter` to consider the maximum.

dccote, 2020-05-29:
I think this particular case is:
1. System2f(10, 100_000) has no aperture stop or field stop
2. chief ray, when no y is provided, uses the edge of the field of view, which in this case is infinite.

In chiefRay, we should check if fieldOfView is defined or infinite before setting y to it:
```
        if y is None and isfinite(self.fieldOfView()):
            y = self.fieldOfView()
        else:
            raise ValueError("Must provide y when field of view is infinite")
```

commit ead0c80d (2020-05-29): Fix #193
PR #200 (2020-05-29): ChiefRay raises exception when no y and infinite field of view
When we create an `ImagingPath` with an infinite field of view and we don't specify `y`, we get a ray with y = inf and theta = -inf. Now, if the field of view is infinite and we don't give a specific `y`, an exception is raised.

Fixes #193
closed 2020-05-29 by dccote