DCC-Lab/RayTracing #198: Readthedocs: contributors should be highlighted on main page

dccote, 2020-05-29:
I think it is now time to declare that raytracing is not Daniel Côté's project, but the DCCLab project, with names of contributors.  I would like that to appear directly on the main entry page: https://raytracing.readthedocs.io/en/latest/

Also, we may keep my email, but the best way to mention bugs should be to write an issue (anyone! even people from outside the group). We should say that somewhere and explain how to do that.

ElaheParham, 2020-05-29:
I will revise the first page and add a how to report bugs (making issue)

dccote, 2020-05-29:
Let's name everyone involved (full names):
Elahe, Valerie, Gabriel, etc....

ElaheParham, 2020-05-30:
Not everybody has written their full name in their GitHub accounts!
I used the ones that I knew and the ones that were written on their Git account.
Can you give me a list to make sure nobody is forgotten?

closed 2020-05-31 by dccote