DCC-Lab/RayTracing #216: The chiefRay does not always allign with the images

JLBegin, 2020-06-01:
When using `onlyChiefAndMarginalRays=True`. The top of the arrow should fit with the chief Ray. It seems to have something to do with conjugates calculations. 

example : 

```
path = ImagingPath()
path.label = "Thick diverging lens"
path.objectHeight = 20
path.append(Space(d=50))
path.append(ThickLens(R1=-20, R2=20, n=1.55, thickness=10, diameter=25, label='Lens'))
path.append(Space(d=50))
path.display(onlyChiefAndMarginalRays=True)
```
[image]
[image]

dccote, 2020-08-25:
@JLBegin This code produces images that look weird.
[image]

JLBegin, 2020-08-25:
What is weird? The size of the FoV graphics? Or the spherical interface graphic not fitting with the rays?

dccote, 2021-01-07:
In fact, now that I think about it, the green line does NOT have to align with the image.  It is the continuity of the other green line that should, and does.

closed 2021-01-07 by dccote