DCC-Lab/RayTracing #224: LaserPath.display()

ElaheParham, 2020-06-03:
I was trying to run an example with the Laser path.display. I have the following example:

```
from raytracing import *
# define the elements in the optical path
Spc1=Lens(f=10,label='Space1') # Space d=10
Lens1=Space(d=10,label='Lens1') # Lens f=10
Spc2=Lens(f=10,label='Space2') # Space d=20
#define a gaussian beam
beam=GaussianBeam(q=complex(1,1))
pathL=LaserPath(elements=[Spc1,Lens1,Spc2])
pathL.display(inputBeam=beam)
```

and it raises the error:
> ypeError: 'int' object is not callable

and it refers to the function: laserpath.createBeamTracePlot() line 130:

> displayRange = 2 * self.largestDiameter()

When I rewrite it like : `displayRange = 2 * self.largestDiameter` there is no more error.
can you please review this function?

JLBegin, 2020-06-03:
We forgot to update the lines a few days ago when largestDiameter was turned into a property. I fixed the call in master.

closed 2020-06-03 by JLBegin

ElaheParham, 2020-06-03:
Merci, I will fix that in my branch either :)