DCC-Lab/RayTracing #229: Laserpath.eigenMode() valid values for radii of curvature

ElaheParham, 2020-06-03:
I cannot understand the application of this function. I wrote an example using that:

```
# define the elements in the optical path
Lens1=Lens(f=5,label='Lens1') # Lens f=10
Spc1=Space(d=2,label='Space1') # Space d=10

#define a gaussian beam
beam=GaussianBeam(q=complex(2,2))

pathL=LaserPath(elements=[Spc1,Lens1,Spc1,Spc1,Lens1,Spc1])
pathL.inputBeam=beam
print(pathL.eigenModes()[0])
print(pathL.eigenModes()[1])
```

The output is:
```
Not valid complex radius of curvature
Complex radius: (-2.31e-16+4j)
w(z): 0.028, R(z): -69175290276410816.000, z: 0.000, λ: 632.8 nm
zo: 4.000, wo: 0.028, wo position: 0.000 
```

This is the path: Spc1, Lens1, Spc1 (and for the round trip I have duplicated that in the laser path.)
Whatever input I select, one of the outputs (sometimes both), I get 

> Not valid complex radius of curvature

Is there any limitation for the elements in Laserpath or the input Gaussian beam?

dccote, 2020-06-11:
Ok, I did it and I figured it out: there are a few things here:
1. When you plan to calculate the Eigenmodes, you don't need to set `inputBeam` because that is what you are looking for. That's not why this is happening though.
2. `eigenModes` are really coming out of a relatively simple calculation where we look for the q = (Aq + B)/(Cq+D), so it is a very simple quadratic equation and you always get two answers. So here, the function does return the two values of q that satisfy the equation.
3. However, 1/q = 1/R - j lambda/pi w^2, so if the imaginary component is not negative, then we don't have a gaussian, we have an unphysical q that must be dropped. *I forgot to code that*.

So, in eigenModes, we simply need to return the solutions, but only if they are physically meaningful. There is a @property called `isFinite` that can be used just for that.  As an exercise, I'll let you do it... :-)

ElaheParham, 2020-06-11:
Thank you.
That was clear to me

dccote, 2020-06-14:
Nothing to fix, proper behaviour.

closed 2020-06-14 by dccote