DCC-Lab/RayTracing #230: Changing object position doesn't change the ray tracing.

Valeriouus, 2020-06-03:
Setting the object position changes the actual position of the object when display(), but the rays still start from position 0 as if the object was still there. 
[image]

illumination3 = ImagingPath()
illumination3.objectPosition = 80
illumination3.objectHeight = 5
illumination3.fanNumber = 1
illumination3.fanAngle = 0
illumination3.design['rayColors'] = ['r', 'r', 'r']

illumination3.append(Space(d=10))
illumination3.append(Lens(f=10, diameter=100, label="Collector"))
illumination3.append(Space(d=10+30))
illumination3.append(Lens(f=30, diameter=100, label="Condenser"))
illumination3.append(Space(d=30+30))
illumination3.append(Lens(f=30, diameter=100, label="Objective"))
illumination3.append(Space(d=30+30))
illumination3.append(Lens(f=30, diameter=100, label="Tube"))
illumination3.append(Space(d=30+30))
illumination3.append(Lens(f=30, diameter=100, label="Eyepiece"))
illumination3.append(Space(d=30+2))
illumination3.append(Lens(f=2, diameter=10, label="Eye Entrance"))
illumination3.append(Space(d=2))
illumination3.display()

Valeriouus, 2020-06-03:
#152

dccote, 2020-06-11:
In fact, objectPosition was never used properly and should be removed, or used properly with object.  I am not sure if it is that useful.  Could this be done manually by changing `Space()`?

Valeriouus, 2020-06-11:
It is useful for kohler illumination example. I want to design a system and put the sample and the light source in this system to see both their illumination paths. However, the sample won't be at the beginning of the path, but in between the condenser and the objective, and the light source will be at the beginning. 
objectPosition would be a nice feature of the module, but there might be another way of doing this.

closed 2020-07-21 by dccote