DCC-Lab/RayTracing #238: Example : Laser Scanning Microscopy

Valeriouus, 2020-06-06:
Here is my code for now. 

import envexamples
from raytracing import *

# A polygonal mirror with 36 facets has an angle of 10 degrees, 0.1750 rad, between the facets.

class LUMPlanFL40X(Objective):

    def __init__(self):
        super(LUMPlanFL40X, self).__init__(f=180/40,
                                         NA=0.8,
                                         focusToFocusLength=40,
                                         backAperture=7,
                                         workingDistance=2,
                                         label='LUMPlanFL40X Objective',
                                         url="https://www.edmundoptics.com/p/olympus-lumplfln-40xw-objective/3901/")

illumination = ImagingPath()
# Let's say that the laser has a diameter of 500 nm.
illumination.objectHeight = 0.0005
illumination.rayNumber = 3
illumination.fanNumber = 3
illumination.fanAngle = 0
illumination.append(System4f(f1=40, f2=75, diameter1=24.5, diameter2=24.5))
illumination.append(System4f(f1=100, f2=100, diameter1=24.5, diameter2=24.5))
illumination.append(Space(d=180/40))
illumination.append(LUMPlanFL40X())
illumination.append(Space(d=180/40))
illumination.display()

- I need to change the angle of which the rays are propagating at the object plane
- I would like to make a list of the image position in y for each object angle.

shadimasoumi, 2020-06-11:
The text is done!

Valeriouus, 2020-06-30:
The text is done in the article. The code is done in the article and in the _examples_ folder. Please tell me your comments! :)

Valeriouus, 2020-07-08:
I deleted the previous graph and kept Daniel's changes. Also, the text has been changed in the article. Plz review!

closed 2020-07-28 by shadimasoumi