DCC-Lab/RayTracing #240: How can I make a list of the image position in y?

Valeriouus, 2020-06-06:
For example Laser scanning microscopy, every time the angle of the object is change, I need to add a position of the image in y in a list. How can I obtain that info? See #238 and #239.

JLBegin, 2020-06-08:
you can use intermediateConjugates()
```
path = ImagingPath()
...
planes = path.intermediateConjugates()
```
This function calculates the position and the magnification of the conjugate planes.
Returns the list of position and magnification of conjugate planes. The first index is the position of the conjugate plane and the second index is the magnification

Valeriouus, 2020-06-08:
Doesn't work for this example, but is very useful for _confocaldetection.py_. I don't want the position in z, but the height of the image for a certain object angle.

closed 2020-06-10 by dccote