DCC-Lab/RayTracing #251: Principal ray should appear as the chief ray passing through the FS with the highest height.

Valeriouus, 2020-06-11:
def imagingPath(a=10, b=10, title=""):
    
    path = ImagingPath()
    path.label=title
    path.append(System4f(f1=50, diameter1=a, f2=50, diameter2=b))
    path.append(Aperture(diameter=10, label='Camera'))
    
    return path

path = imagingPath(a=15, b=15)
path.display(limitObjectToFieldOfView=False, onlyChiefAndMarginalRays=True)

[image]

The green ray should pass just at the edge of the FS, even if it doesn't start at the top edge of the object. Then, it's obvious that not all the object will be properly imaged.

dccote, 2020-06-11:
As discussed, because you set the object manually, the edge of the object is not the principal ray.   This incoherent option has been removed from display().

dccote, 2020-06-11:
Reopened after discussion with @Valeriouus : we want to draw the chief ray properly, regardless of what we use for the object.

closed 2020-06-12 by dccote