DCC-Lab/RayTracing #276: Materials should take wavelength as a optical parameter

dccote, 2020-06-16:
We should have a wavelength for a ray, and use it as a parameter when getting the index of refraction of materials. That way, we could fully characterize the chromatic aberrations for a given doublet.

dccote, 2020-06-19:
@ElodieFlora would like to contribute. Are you planning to do it soon or could she contribute?

PyMarc2, 2020-06-19:
Oh well I planned working on it this weekend, but I'd prefer if she did it if she wants to get involved! :-) @ElodieFlora

dccote, 2020-06-19:
Great: you can both work on it, @PyMarc2 you can review her work (then I will complete the merge).

PyMarc2, 2020-06-19:
Perfect, I'll help!

PyMarc2, 2020-07-11:
This branch is at the testing step. I'll be showing Elodie and Charels-Eric how to test properly with the unitesting package. Probably towards the begining of next week.

closed 2020-07-16 by dccote