DCC-Lab/RayTracing #279: demo #3-13-16-18-19 not working with 1.2.10

AntoineRousseau97, 2020-06-18:
I installed raytracing with pip.
i called "python -m raytracing"
demo 1 and 2 worked, but not 3, so others didnt display.
i called every demo 1 by 1 to see which one is working.
It looks like some demos are not displaying correclty (i think).

bopaquin, 2020-06-18:
13 is not an error, it's just not a user friendly display:
```
(10.0, <raytracing.matrix.Matrix object at 0xFFFFFFFFFFFFFFFF>)
(-10.0, <raytracing.matrix.Matrix object at 0xFFFFFFFFFFFFFFFF>)
```
I don't know if it is intended.

JLBegin, 2020-06-18:
I'm looking at this on branch `fixDemos`.
- we now default to onlyPrincipalAndAxialRays. I use to check if there was an ImagingPath.apertureStop() but this is not really enough since we can still have a PrincipalRay = None so I added this other check since I don't think we want to only show axialRay. 
- The GraphicOf was missing the call for ThickLens. Fixed.

I still have to readjust the displayRange, it's not always ideal.

PR #280 (2020-06-22): Fix display of demos
fix #279

- Block the use of 'onlyPrincipalAndAxialRays' if ImagingPath.principalRay() is None
- Fixed ThickLens graphic
- Fixed many displayRange and plt scaling issues

#### update
- Fixed graphic for subclasses of specialty lenses
- Added display method in Matrix() which calls Graphic(self) to display any Matrix or MatrixGroup subclasses correctly.
closed 2020-06-22 by dccote