DCC-Lab/RayTracing #295: Actual objectHeight should set self.fieldOfView() correctly in ImagingPath().

Valeriouus, 2020-06-25:
Here is an example of the bug : 

import envexamples
from raytracing import *

path = ImagingPath()
# The object is a 5 mm light source propagating light with a NA of 24 degrees
path.objectHeight = 5
path.fanAngle = 0.41888
# The light propagates through a 4f system. The magnification is f2/f1 = -3.75. 
path.append(System4f(f1=40, diameter1=25.4, f2=150, diameter2=50.8))

# Verification of the magnification of the system with different methods. 
# Method 1 : imageSize()
# Should give 5 * 3.75 = 18.75 mm. 
print("Image size is {}".format(path.imageSize()))

# Method 2 : Finding the intermediate conjugates.
# intermediateConjugates() is a dictionnary of the position and magnification of all conjugate planes of the object.
planes = path.intermediateConjugates()  
# The last conjugate plane is the image plane. The magnification of this position is saved in mag. 
mag = planes[-1][1] 
print("Magnification {}".format(mag)) # Should give -3.75

# Verification of the size of the FOV used in imageSize(). 
FOV = path.fieldOfView()
print("FOV is {}".format(FOV)) # Should be 5 mm since path.objectHeight = 5.

path.display(onlyPrincipalAndAxialRays=False)

dccote, 2020-06-25:
This is interesting, we do have a "problem" with nomenclature: in an optical system, the *image size* is always calculated from the *size of the field of view*. I don't think there is an accepted name of "maximal image size", although I understand the idea. 

The problem comes from the fact that we accept an `objectHeight` as a parameter, and quite obviously for the user, `imageSize` should be the size of the object times magnification. I am hesitating between two options:

1. Modify `imageSize()` to consider `objectHeight`, but then *image size* will not mean what people might expect, because it should really be the FOV * Magnification
2. Remove `objectHeight` altogether from the `ImagingPath`, but add an argument to most functions depending on object size, where the user can add the object height wanted. If nothing is passed as a parameter, then use the field of view.  I know, this is extreme.

I don't know what to do. @GabGabG @Valeriouus @JLBegin @shadimasoumi @ElaheParham

dccote, 2020-06-26:
Ok, I studied the code: `objectHeight` is really only used for drawing. It is not used in the code of `ImagingGroup`.  We could remove it completely from `ImagingPath`, and pass it as a parameter to `display()`. This would simply a few things because if we don't pass an objectHeight, we would take the field of view, it would be expected from the user.  However, this will require some changes because many examples set the `objectHeight`, and now we would need to deprecate it.  I think storing it and then warning the user is a good option, we can then pass it to `display()` as a temporary measure.

I will create a branch to experiment and see what we get.

Valeriouus, 2020-06-26:
I'm not sure this is something you had in mine, but I definitely think that it should be possible for the user to be able to set a certain objectHeight if necessary. For example, for my setup, I know exactly the size of the object (the LSR has a diameter of 5 mm) and I would use the module to determine which 4f system to use to get the beam bigger so it fits the objective BA. 
Perhaps the user should be able to set the fieldOfView in this case? The module says if it fits or not depending on the chosen 4f system? I'm not sure...

dccote, 2020-07-16:
Ludo is working on it, where object and field of view are really treated independently.  I'm closing this issue since handled with #318

closed 2020-07-16 by dccote