DCC-Lab/RayTracing #302: style of ABCD matrix

ElaheParham, 2020-06-30:
Hi,
Can we change the printing style of ABCD matrix?

It is good, but when I try to run doctest, it is confusing and most of the matrix prints will be failed.
If you want to run it to see what I mean you can use the following code:
```

import doctest
import raytracing

doctest.testmod(m=raytracing.axicon,verbose=True)
doctest.testmod(m=raytracing.components,verbose=True)
doctest.testmod(m=raytracing.eo,verbose=True)
doctest.testmod(m=raytracing.figure,verbose=True)
doctest.testmod(m=raytracing.gaussianbeam,verbose=True)
doctest.testmod(m=raytracing.imagingpath,verbose=True)
doctest.testmod(m=raytracing.lasercavity,verbose=True)
doctest.testmod(m=raytracing.laserpath,verbose=True)
doctest.testmod(m=raytracing.materials,verbose=True)
doctest.testmod(m=raytracing.matrix,verbose=True)
doctest.testmod(m=raytracing.matrixgroup,verbose=True)
doctest.testmod(m=raytracing.ray,verbose=True)
doctest.testmod(m=raytracing.rays,verbose=True)
doctest.testmod(m=raytracing.specialtylenses,verbose=True)
doctest.testmod(m=raytracing.utils,verbose=True)

```

dccote, 2020-07-07:
I think we definitely can and probably should. What do you suggest?

ElaheParham, 2020-07-07:
Probably something like : 
A=a, B=b, C=.... can be less miss written when we prin it.
but if you prefer a matrix form it can be in a simple form like
|A B|
|C D|
or {[A,B],[C,D]}
because I think using \ and / and a lot of the blank spaces is not good.

PyMarc2, 2020-07-16:
I agree that A=, B=, C=, D= is less confusing and more concise. 
But I could understand the purpose of having it as a matrix as well...

PyMarc2, 2020-07-16:
Maybe something like this? Both worlds.

|   A=12.31231     B=32423    |
|   C= 32.0002      D=0.3214   |

ElaheParham, 2020-07-16:
> I agree that A=, B=, C=, D= is less confusing and more concise.
> But I could understand the purpose of having it as a matrix as well...

I can understand the purpose of having a matrix style too, but there are two problems for now.
first is with backslashes in the docstring.
and Second, having <blank> lines between the printed matrix and the printed focal length.

when I want to use doctest to test the examples they real output does not match with what I have put in the examples, so it raises some error.

dccote, 2020-07-16:
I'm fine without the matrix.  Let's remove it. Same with ray.

ElaheParham, 2020-07-17:
I will make a branch from master and test some styles

dccote, 2020-07-17:
Done #321