DCC-Lab/RayTracing #316: Making the font size of axis numbers, legends and labels bigger.

shadimasoumi, 2020-07-14:


JLBegin, 2020-07-14:
I agree, but It might depend on the machine. Here is an example of the current display for demo #5.
Can you check that you have a similar display on your end? I also understand that we have to check all display types (ImagingPath, GaussianBeam, Intensity profiles)

```python
path = ImagingPath()
path.label = "Demo #5: Simple microscope system"
path.append(Space(d=4))
path.append(Lens(f=4, diameter=0.8, label='Obj'))
path.append(Space(d=4 + 18))
path.append(Lens(f=18, diameter=5.0, label='Tube Lens'))
path.append(Space(d=18))
path.display()
```
[image]

shadimasoumi, 2020-07-14:
This is mine!

[image]

closed 2020-07-27 by dccote