DCC-Lab/RayTracing #338: Error when calling display()

gordon13, 2020-08-11:
I've installed the latest version from github, and the following code fails at the display() part.

This is version 1.2.12 of the package, and running on Python 3.7.

Here is the code I am trying to run:
```
from raytracing import *

path = ImagingPath()
path.append(Space(d=100))
path.append(Lens(f=50, diameter=25))
path.append(Space(d=120))
path.append(Lens(f=70))
path.append(Space(d=100))
path.display()
```

Here is the result:
```
Traceback (most recent call last):
  File "C:\users\paul\documents\projects\irguard\optical_35mm_lens.py", line 505, in <module>
    path.display()
  File "C:\Users\Paul\Miniconda3\lib\site-packages\raytracing-1.2.12-py3.7.egg\raytracing\imagingpath.py", line 919, in display
  File "C:\Users\Paul\Miniconda3\lib\site-packages\raytracing-1.2.12-py3.7.egg\raytracing\figure.py", line 536, in display
  File "C:\Users\Paul\Miniconda3\lib\site-packages\raytracing-1.2.12-py3.7.egg\raytracing\figure.py", line 604, in display2D
  File "C:\Users\Paul\Miniconda3\lib\site-packages\raytracing-1.2.12-py3.7.egg\raytracing\figure.py", line 634, in draw
  File "C:\Users\Paul\Miniconda3\lib\site-packages\raytracing-1.2.12-py3.7.egg\raytracing\figure.py", line 719, in updateLabels
  File "C:\Users\Paul\Miniconda3\lib\site-packages\raytracing-1.2.12-py3.7.egg\raytracing\figure.py", line 749, in fixLabelOverlaps
  File "C:\Users\Paul\Miniconda3\lib\site-packages\raytracing-1.2.12-py3.7.egg\raytracing\figure.py", line 738, in getRenderedLabels
  File "C:\Users\Paul\Miniconda3\lib\site-packages\raytracing-1.2.12-py3.7.egg\raytracing\graphicComponents.py", line 413, in isRenderedOn
AttributeError: 'Text' object has no attribute 'get_tightbbox'
```

Any ideas?

dccote, 2020-08-16:
With the current master, this appears to work fine. Can you update and confirm?

dccote, 2020-08-16:
@gordon13 Can I see the `optical_35mm_lens.py` file?

PR #351 (2020-08-27): matplotlib 3.0 or above is required
Required version to fix #338
closed 2020-08-27 by dccote