DCC-Lab/RayTracing #355: onlyChiefAndMarginalRays renamed to onlyPrincipalAndAxialRays?

calocedrus, 2020-09-09:
I'm running the `demo.py` and it stops at the Thick diverging lens example complaining that the keyword `onlyChiefAndMarginalRays` is unexpected in the call to `path.display(onlyChiefAndMarginalRays=True)`.
Tracing back in the old code I had (raytracing 1.1.9) and the current (latest, from master branch) I see that there is no more `onlyChiefAndMarginalRays` in `raytracing/imagingpath.py` , it seems it's been renamed to `onlyPrincipalAndAxialRays`, is that correct?
Ludo

dccote, 2020-11-20:
Yes it is.  We will fix the files, I believe we had forgotten the `demo.py` file.

dccote, 2021-01-06:
Done.  It is in the `1.3.1-bugFixes` branch and will be merged once the article is out and other bugs have been found and fixed.

closed 2021-01-06 by dccote