DCC-Lab/RayTracing #363: Objectives not displaying and new arguments missing

FrancoisCoteIng, 2021-01-08:
I wanted to display a Nikon objective using the specialityLenses class "Objective". It seems that there are 2 new arguments needed to define an objective: Field Number (which is not specified by all manufactures) and magnification (which depend on the tube lens used, so I don't see the point to ask for it if the focal is already in the definition). Even if I add those new arguments, it seems that the algorithm doesn't want to draw my objective and can't flip it using .flipOrientation().

dccote, 2021-01-08:
The guilty will be found and hanged.

dccote, 2021-01-08:
@FrancoisCoteIng We need example code to reproduce this, since the `python -m raytracing` examples work fine.

FrancoisCoteIng, 2021-01-08:
try:
    from raytracing import *
except ImportError:
    raise ImportError('Raytracing module not found: "pip install raytracing"')

import nikonObj as nikon

class Confocal:
    @staticmethod
    def detection():
        detection = ImagingPath()
        detection.label = "VMS/Confocal Base detection"
        detection.objectHeight = 0.5+0.2 # mm maximum, include diffuse spot size
        detection.fanAngle = 1.1 # NA = 1.1
        detection.fanNumber = 11
        detection.rayNumber = 3
        detection.showImages = False

        fTubeLens=200
        fScanLens=54
        f1=150
        f2=75

        obj = nikon.N25ApoMD()
        obj.flipOrientation()

        detection.append(Space(d=2))
        detection.append(obj)
        detection.append(Space(d=fTubeLens)) 
        detection.append(Lens(f=fTubeLens,diameter=50, label='Tube Lens'))
        detection.append(Space(d=fTubeLens))
        detection.append(Space(d=fScanLens))
        detection.append(Lens(f=fScanLens, diameter=30, label='Scan Lens'))
        detection.append(Space(d=fScanLens))

        return detection

if __name__ == "__main__":

    test = Confocal.detection()
    test.showImages = True
    test.objectHeight = 1
    test.rayNumber = 5
    test.fanNumber = 5

    test.display()

FrancoisCoteIng, 2021-01-08:
nikonObj.py :

from raytracing.specialtylenses import *

class LWD16X(Objective):
    """ Nikon 16x immersion objective

    .. csv-table::
        :header: Parameter, value

        "f", "200/16"
        "backAperture", "20"
        "Numerical Aperture (NA)", "1.56"
        "Working Distance (mm)", "3"

    Notes
    -----
    Immersion not considered at this point.
    """

    def __init__(self):
        super(LWD16X, self).__init__(f=200/16,
                                         NA=1.95*0.8,
                                         focusToFocusLength=75,
                                         backAperture=20,
                                         workingDistance=3,
                                         label='CFI75 LWD 16x W',
                                         url="https://www.thorlabs.com/thorproduct.cfm?partnumber=N16XLWD-PF")

class N25ApoMD(Objective):
    """ Nikon 25x immersion objective

    .. csv-table::
        :header: Parameter, value

        "f", "200/25"
        "backAperture", "17.6"
        "Numerical Aperture (NA)", "1.1"
        "Working Distance (mm)", "2"

    Notes
    -----
    Immersion not considered at this point.
    """

    def __init__(self):
        super(N25ApoMD, self).__init__(f=200/25,
                                         NA=1.1,
                                         magnification=25,
                                         fieldNumber=20,
                                         focusToFocusLength=76,
                                         backAperture=17.6,
                                         workingDistance=2,
                                         label='N25 APO MD',
                                         url="https://www.thorlabs.com/thorproduct.cfm?partnumber=N25X-APO-MP")

JLBegin, 2021-01-22:
While I am addressing the graphic issues on #365, I am not sure about the arguments issue. 

Here is our current definition for an objective. We are indeed specifying focal length and magnification seperatly, leaving the tubelens to be implicitly defined. A rework of that could be considered, but an API change is a delicate matter... @dccote 
[image]

dccote, 2021-01-27:
> While I am addressing the graphic issues on #365, I am not sure about the arguments issue.
> 
> Here is our current definition for an objective. We are indeed specifying focal length and magnification separately, leaving the tubelens to be implicitly defined. A rework of that could be considered, but an API change is a delicate matter... @dccote

Yes, no API change.

However, let's compute the magnification or tube lens or whatever and confirm that they are coherent.  If they are not, we raise an exception.

closed 2021-02-15 by dccote