DCC-Lab/RayTracing #366: If no  field stop then reportEfficiency() fails with

dccote, 2021-01-27:
This code fails at `reportEfficiency()`:

```
from raytracing import *

# On construit le chemin optique a partir de la lampe
f = 50
path = ImagingPath()
path.append(Space(d=f))
path.append(Lens(f=f,diameter=100))
path.append(Space(d=f))

# On calcule le profil d'intensité en propageant plein de rayons
# qui suivent la distribution en cos^2 theta du probleme.
nRays = 100000 # Vous pouvez augmenter pour obtenir une meilleure reponse
inputRays = RandomLambertianRays(yMax=0.0001,maxCount=nRays)

outputRays = path.traceManyThrough(inputRays)
efficiency = 100*outputRays.count/inputRays.count
path.display()
outputRays.display("Output profile {0:.0f}% efficiency".format(efficiency), showTheta=False)
path.reportEfficiency()
```

JLBegin, 2021-01-27:
Should we raise an exception saying that we require a field stop to compute the efficiency or there's a way to reportEfficiency without it?

dccote, 2021-05-05:
> Should we raise an exception saying that we require a field stop to compute the efficiency or there's a way to reportEfficiency without it?

I think efficiency without a field stop or aperture stop will always be 100%, but I am not sure. We should add a function `hasFieldStop` and `hasApertureStop`.

dccote, 2021-05-07:
I worked on this, and if we don't have a principal ray, then we are somehwat screwed. For that, we need a FS.  Without an FS, we can't report eaningful things, so I print that and return.

closed 2021-05-12 by dccote