DCC-Lab/RayTracing #367: Graph fails if no field stop

dccote, 2021-01-27:
This code prints out a warning in the graph code:

```shell
/Users/dccote/anaconda3/lib/python3.7/site-packages/raytracing/figure.py:109
UserWarning:Infinite field of view: cannot use limitObjectToFieldOfView=True.

/Users/dccote/anaconda3/lib/python3.7/site-packages/raytracing/figure.py:116
UserWarning:Infinite image size: cannot use limitObjectToFieldOfView=True.
```

with this code on most recent 

```python
from raytracing import *

# On construit le chemin optique a partir de la lampe
f = 50
path = ImagingPath()
path.append(Space(d=f))
path.append(Lens(f=f,diameter=100))
path.append(Space(d=f))

# On calcule le profil d'intensité en propageant plein de rayons
# qui suivent la distribution en cos^2 theta du probleme.
nRays = 100000 # Vous pouvez augmenter pour obtenir une meilleure reponse
inputRays = RandomLambertianRays(yMax=0.0001,maxCount=nRays)

outputRays = path.traceManyThrough(inputRays)
efficiency = 100*outputRays.count/inputRays.count
path.display()
outputRays.display("Output profile {0:.0f}% efficiency".format(efficiency), showTheta=False)
path.reportEfficiency()
```

JLBegin, 2021-01-27:
I think the warnings are fine since your object is at focal length. The code seems to run well except for the efficiency report which fails without field stop.

dccote, 2021-01-27:
Let's warning: "Warning: blablabla" instead of UserWarning, which sounds too technical for no reason.
In the warning, we should say what corrective measure we took: limite size to arbitrary number, etc... so the user knows what we have done and how they could fix it.