DCC-Lab/RayTracing #374: No control over rayNumbers, fanNumbers and fanAngle

FrancoisCoteIng, 2021-02-17:
I can't control the fanAngle, fanNumber and rayNumbers on my traces. I think a checkbox is supposed to be present in the display labeled "ObjectRays" but doesn't show. I always seem to have 3 rays (top, bottom and middle position of the object), no matter what parameters I fix for the rays.

try:
    from raytracing import *
except ImportError:
    raise ImportError('Raytracing module not found: "pip install raytracing"')

class Confocal:
    @staticmethod
    def detection():
        detection = ImagingPath()
        detection.label = "VMS/Confocal Base detection"
        detection.objectHeight = 10 # mm maximum, include diffuse spot size
        detection.fanAngle = 1.1 # NA = 1.1
        detection.fanNumber = 5
        detection.rayNumber = 5
        detection.showImages = False

        fTubeLens=200
        fScanLens=54
        f1=150
        f2=75

        detection.append(Space(d=fTubeLens)) 
        detection.append(Lens(f=fTubeLens,diameter=50, label='Tube Lens'))
        detection.append(Space(d=fTubeLens))
        detection.append(Space(d=fScanLens))
        detection.append(Lens(f=fScanLens, diameter=30, label='Scan Lens'))
        detection.append(Space(d=fScanLens))

        return detection

if __name__ == "__main__":

    test = Confocal.detection()

    test.display()

dccote, 2021-02-22:
This has changed since 1.2.x: you must now use:
`test.display(rays=myRays)`
where you can pass any type of rays you want (it become too complicated to manage all the options directly in `ImagingPath`).  You would use: `myRays = UniformRays(yMax=1, thetaMax=0.5, M=3, N=3)` or any other `Rays` subclass (at https://raytracing.readthedocs.io/en/1.3.1/reference.html#rays)

so your code can be modified to be:

```
    test = Confocal.detection()
    myRays = UniformRays(yMax=5, thetaMax=0.5, M=5, N=5)

    test.display(rays=myRays)
```

JLBegin, 2021-02-24:
The old API for object display is still supported so Francois' code has to work.

JLBegin, 2021-02-24:
The old API for object display is still supported, so Francois' code still works. Since no ObjectRays are provided, a default ObjectRays is created with Francois' settings and shown in the display with `Object/Image` checkbox. 

And it is "working" but the problem comes from the fanAngle that is set too high for the outer rays to go in the system, and that by default `removeBlocked` (rays) is set to True. So we are getting fooled by the display. Calling `test.display(removeBlocked=False)` shows what's going on:

[image]

I understand this might not be ideal. We could
- Print a warning when some rays are getting blocked. 
- Set `removeBlocked` to False by default.

JLBegin, 2021-02-24:
But then I also noticed a bug: the default ObjectRays created from ImagingPath attributes does not set the fanNumber (number of fans). So it is always 3. I will fix this independantly from this issue.

dccote, 2021-03-02:
Maybe the default parameters should be computed better. the fanAngle for instance could be estimated better to fit.

closed 2021-03-10 by dccote