DCC-Lab/RayTracing #403: imageSize does not behave as expected for beginners versus advanced users

dccote, 2021-05-05:
The function `imageSize` of `ImagingPath` returns the size of the image of the field of view: it is the accepted definition in a real optical system, because the limit really is the field of view, not the object. However, in a very simple system (say a single lens with an infinite diameter), the field of view is infinite, because you need at least two finite diameters to get a finite field of view (i.e. a field stop). The following code:
```
from raytracing import *
path = ImagingPath()
path.objectHeight = 10
path.append(Space(d=100))
path.append(Lens(f=50))
print("Image size is: {0}".format(path.imageSize()))
```

will print:
`Image size is: inf`

because the FOV in this example is infinite. This unfortunately, while being correct and in line with the documentation and the definition, appears really odd to anyone who is not an expert.

After discussion, we believe the appropriate behaviour should be:

1. `imageSize` will accept an argument `useObject=False`. In this case, it will 
2. If the FOV is infinite, the module will print a warning to tell the user to use `useObject=True` to get the expected answer.
3. We are not sure if `maxImageSize` would be an approriate function to add (it would always use the FOV), but we think this changes the semantic (what becomes the meaning of `imageSize` then?)

ElaheParham, 2021-05-05:
So I think the argument Name can be useObjectHeight which is optional and can be True or False, so the algorithm can be like:

If no input parameter is defined or it is False : 
        Then use FOV (Here we will have two cases)
                If FOV is infinite : 
                        Then print +Inf and a warning (this is because FOV is Inf and if you mean to use object height pay
                         attention to the input argument!)

if useObjectHeight is True then just use object height for calculation.

*********

It can be noted in the documentation that having an infinite FOV is not something that happens in
real experiments but we have considered this case for educational purposes.

dccote, 2021-05-05:
The branch `imageSizeScrewup` implements this.

closed 2021-05-05 by dccote