DCC-Lab/RayTracing #429: f/# in reportEfficiency() = inf or NaN.

Valeriouus, 2021-11-01:
**1. Example with f/# = inf.** 
```
from raytracing import *

# defines the collection path, considering full image size and full NA.
path = ImagingPath()
path.append(olympus.XLUMPlanFLN20X().flipOrientation())
path.append(Space(d=180))
path.append(Lens(f=180, diameter=50.8, label="Tube lens"))
path.append(Space(d=180))
path.append(Aperture(diameter=18.83, label="Camera"))
print(path.entrancePupil())
path.displayWithObject(diameter=1.1, fanNumber=10, rayNumber=5, fanAngle=1, removeBlocked=False)
```
[image]

We think the f/# equals inf because the focal length of a telescope is inf. Raytracing automatically places the object at inf for a telescope (?), but I know that I want to place the object at the focal length of the first lens (here being the objective). Therefore, the f/# should be (focal length of objective)/(diameter of entrance pupil) = 9mm/4mm = 2.25. 

**2. Example with f/# = NaN.**
```
from raytracing import *

path=ImagingPath()
path.append(Space(d=10))
path.append(Lens(f=10))
path.append(Space(d=10.01))
path.append(Aperture(diameter=2))
print(path.entrancePupil())
path.display()
```
[image]

Because the AS is placed exactly at the focal plane of the lens, the entrance pupil is at inf and has a diameter of in _because raytracing puts the object af inf_ again. I know that the object must be at the object plane of the lens, so at the focal plane on the right side of the lens, so the f/# should be again (focal of lens)/(diameter of entrance pupil). Not sure how to find the diameter of the entrance pupil here though.

PyMarc2, 2021-11-01:
we could add a objectAtFocal=True in the report function, so that we just change the object position for the calculation ? Thoughts ?

mirequem, 2021-11-02:
The two systems shown above are telecentric systems: the stop aperture is at the focal point, so the entrance pupil is at infinity. In this situation, we have an indeterminate f-number.

According to Smith: the f-number and the numerical aperture are two quantities that define the same characteristic of a system. In general, one uses:
f-number: infinite conjugation systems ;
NA: finite conjugation systems.

In most telecentric systems, finite conjugation is used: the object is rarely placed at infinity. We will therefore use the following definition for the f-number:

f-number = 1/2NA, with NA = n'sinU', where n' is the refractive index at the image plane and U' is the angle of the illumination cone.

So I think that raytracing should not automatically put the object at inf? 

[image]
f
[image]

[image]

closed 2022-01-10 by github-actions[bot]