DCC-Lab/RayTracing #433: Version check slows down module import in offline environments

himbeles, 2021-11-12:
Thank you for making this nice raytracing library available.
In our corporate environment, we can not rely on a permanent internet connection. 
Consequently, we are getting wait times of several seconds on every import of `raytracing` with failing automated module version check.

Would it be an option for you to remove this automated check? 
Anyhow I think, it is good practice to leave the version control to the package manager and end user. 

Thank you again.

himbeles, 2021-11-13:
PR #434 would remove the automated version check.

dccote, 2021-12-01:
Sorry for the delay.  I really want the automatic version check, I will find a better way of doing it and failing quickly if no internet is available.

himbeles, 2021-12-01:
Sure, thanks for the feedback.

closed 2021-12-01 by himbeles