DCC-Lab/RayTracing #441: 'ImagingPath' object has no attribute 'traceManyTrough'

Carol1117, 2022-05-24:
can not find the function 'traceMany Through' in class ImagingPath

PyMarc2, 2022-05-24:
@Carol1117 
To help us debug faster, please provide the following information:

- Your OS and version
- Your python version
- Your PyTissueOptics module version
- The full code you are trying to run.

dccote, 2022-06-02:
The following code does work:

```
from raytracing import *

path = ImagingPath()
path.append(Space(d=70))
path.append(Lens(f=35, diameter=25))
path.append(Space(d=70))

inputRays = RandomUniformRays()
outputRays = path.traceManyThrough(inputRays)

inputRays.display()
outputRays.display()
```

[intput.pdf](https://github.com/DCC-Lab/RayTracing/files/8826250/intput.pdf)
[output.pdf](https://github.com/DCC-Lab/RayTracing/files/8826251/output.pdf)

closed 2022-07-18 by dccote