DCC-Lab/RayTracing #452: Gaussian Beam: unit inconsistency within print description

Romain-Laine, 2022-08-04:
Hi guys,
First of all, thanks for the really nice tool!
I have started using it to check laser rigs and beam shaping, I found it quite intuitive.

I have noticed an inconsistency in units in the print description for the Gaussian beam:

```
def __str__(self):
    """ String description that allows the use of print(Ray())
    if self.wo is not None:
        description = "Complex radius: {0:.3}\n".format(self.q)
        description += "w(z): {0:.3f}, ".format(self.w)
        description += "R(z): {0:.3f}, ".format(self.R)
        description += "z: {0:.3f}, ".format(self.z)
        description += "λ: {0:.1f} nm\n".format(self.wavelength * 1e9)
        description += "zo: {0:.3f}, ".format(self.zo)
        description += "wo: {0:.3f}, ".format(self.wo)
        description += "wo position: {0:.3f}".format(self.waistPosition)
        return description
    else:
        return "Beam is not finite: q={0}".format(self.q)
```

Since everything is actually defined in mm, the wavelength is also provided in mm and should be printed as:
`description += "λ: {0:.1f} nm\n".format(self.wavelength * 1e6)`

dccote, 2022-08-16:
You are partially correct: the units can be anything you want.  It makes it difficult to set the exact units, but I will find a solution.

Romain-Laine, 2022-08-18:
Thanks, Daniel!
It only looked to me like the constructor expects all units to be in mm (hence the default wavelength being `default=632.8e-6`), but the print has `self.wavelength * 1e9`, effectively printing 632.8nm as 632800nm

dccote, 2022-09-15:
Let me look in more details.

closed 2022-11-22 by github-actions[bot]