DCC-Lab/RayTracing #462: Example code produces error

Slugskickass, 2023-01-18:
When I run the example code which has the option to change the fanAngle I get the following error

UnboundLocalError: local variable 'fanAngle' referenced before assignment

The code I am running is

from raytracing import *
path = ImagingPath() # define an imaging path
#set the desire properties
path.fanAngle = 2
path.objectHeight=5
path.showEntrancePupil = True
path.fanNumber=3
# use append() to add elements to the imaging path
path.append(Space(d=20))
path.append(Lens(f=20,label="f=20"))
path.append(Space(d=30))
path.append(Lens(f=10,label="f=10"))
path.append(Space(d=10))
path.display()

JLBegin, 2023-01-19:
I see this bug was introduced in version 1.3.9 and fixed on master with #457. The next release will include this fix. @dccote 

In the meantime I would recommend to configure the display by providing your desired rays to the display method. In this case, `ObjectRays` would be appropriate. 
```python
from raytracing import *
path = ImagingPath()
path.showEntrancePupil = True

path.append(Space(d=20))
path.append(Lens(f=20,label="f=20"))
path.append(Space(d=30))
path.append(Lens(f=10,label="f=10"))
path.append(Space(d=10))

rays = ObjectRays(diameter=5, halfAngle=2, H=3)
path.display(rays)
```
This is the new and preferred approach. Here halfAngle and H are equivalent to the previous fanAngle and fanNumber, respectively. 

Otherwise you could also downgrade to the previous release 1.3.7 and/or wait for the next release (which should be soon):
`pip install --upgrade raytracing==1.3.7`

dccote, 2023-01-30:
1.3.10 released.  The test code above works.

closed 2023-01-30 by dccote