DCC-Lab/RayTracing #474: Matplotlib update to checkBoxes

JohnKMeyer, 2024-07-29:
I was having an issue running the library with matplotlib 3.9.1, checkBoxes has no attribute rectangles or lines. these have been deprecated in version 3.7 and later.

My workaround was to either install old matplotlib or remove the lines in initVisibilityCheckBoxes that correspond to defining the boxes.

`    def initVisibilityCheckBoxes(self):
        visibility = self.visibility
        if kElementsKey in visibility.keys():
            visibility.pop(kElementsKey)

        subAxes = plt.axes([0.81, 0.4, 0.1, 0.5], frameon=False, anchor='NW')
        self.checkBoxes = CheckButtons(subAxes, visibility.keys(), visibility.values())

        # step = 0.15
        # for i, (label, rectangle, lines) in enumerate(zip(self.checkBoxes.labels,
        #                                                   self.checkBoxes.rectangles,
        #                                                   self.checkBoxes.lines)):
        #     h = 0.85 - step * i
        #     label.set_fontsize(11)
        #     rectangle.set_x(0.05)
        #     rectangle.set_y(h)
        #     rectangle.set(width=0.12, height=0.04)
        #     label.set_y(h + 0.02)
        #     label.set_x(0.2)

        #     lineA, lineB = lines
        #     lineA.set_xdata([0.05, 0.17])
        #     lineB.set_xdata([0.05, 0.17])
        #     lineA.set_ydata([h, h + 0.04])
        #     lineB.set_ydata([h + 0.04, h])

        self.checkBoxes.on_clicked(self.onCheckBoxCallback)
`

dccote, 2024-08-27:
Sorry for the delay and the kerfuffle: I merged the change without testing all examples, and the display of Achromats with their principal planes, BFl, FFL, etc.. was all screwed up, so for now I simply request matplotlib less than 3.8 (excluded) which is a terrible solution, but everything else is still fine.  We will start working on a proper fix immediately.

dccote, 2024-08-28:
Ok, it appears to have been fixed.  A new version will be released later tonight.

closed 2024-08-28 by dccote