DCC-Lab/RayTracing #485: python -m raytracing: error: unrecognized arguments: -a

vstadnytskyi, 2025-04-23:
I am trying "New: A graphical interface for Raytracing"

```
pip install raytracing
python -m raytracing -a
```

I have Mac OS with M2 chip and Python 3.11

`Python 3.11.5 (v3.11.5:cce6ba91b3, Aug 24 2023, 10:50:31) [Clang 13.0.0 (clang-1300.0.29.30)] on darwin`

I get this error

```
> python3 -m raytracing -a
usage: python -m raytracing [-h] [-e EXAMPLES] [-c] [-l] [-t]
python -m raytracing: error: unrecognized arguments: -a
```

JLBegin, 2025-04-23:
It looks like your raytracing package is not up to date. You can update it with the following command:

```
python3 -m pip install --upgrade raytracing 
```

rjsdotorg, 2025-05-05:
I just tried installing  with Win7, py3.8.2 and same result:

 > python -m raytracing -a
 usage: python -m raytracing [-h] [-e EXAMPLES] [-c] [-l] [-t]
 python -m raytracing: error: unrecognized arguments: -a

One problem was that MPL on Win7 stops at 3.7.5

Works fine on Win10 with py3.13

closed 2025-05-12 by JLBegin